turtlebot/turtlebot4 #596: Clear3 Sub-System Not Working

Argon1999, 2025-05-30:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

turtlebot-node and navigation are on 1.0.5
bringup, robot, and diagnostics are on 1.0.3

### Type of issue

Battery

### Expected behaviour

It should be working. I've connected the game controller and it should move. Furthermore I've connected via simple discovery and the keyboard teleop isn't working. 

### Actual behaviour

[image]

The Create3 component is not getting power. The robot is on the dock, and it's saying that it's not on the dock and further suggesting that the battery is critical. Furthermore the mouse and IMU are showing errors that don't tell me anything. I'm assuming this is why it's not moving. 

I tried to run the tests provided and none of them work.
Provided is a picture of the lights that have lit up on the robot itself. I see that the Comm and the Batt light is not on and that the battery percentage at 0.
[image]

### Error messages

```bash

```

### To Reproduce

I'm assuming this is some kind of hardware issue. I don't know how one would reproduce it.

smatarCPR, 2025-06-03:
Hello @Argon1999 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are encountering troubles with your TB4 unit. Based on the information shared I have the following observations/recommendations:

- From the description, it appears that either the Create 3 unit is failing to connect to wi-fi. In simple discover, the Pi and Create3 communicate with each other using wi-fi:

[image]

- If the COMM light is off, this means that Create 3 and Raspberry Pi cannot communicate
- Are you perhaps using a 5.0 GHz network? The Create 3 is unable to connect to 5.0 GHz networks.
- Using the method described here - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi connect to the Raspberry Pi via ethernet and see if you can connect to the Create3 webserve using the address: 192.168.185.3:8080
- Verify the firmware version of the Create3 is the H.2.6, it has been assigned an IP address in the connect tab of the Webserver:

[image]

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

Argon1999, 2025-06-03:
I turned it on and off a few times and it started working.

smatarCPR, 2025-08-27:
Hello @Argon1999 
I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue. 
Best Regards,
Saif

closed 2025-08-27 by smatarCPR