turtlebot/turtlebot4 #617: TurtleBot 4 – Multiple Persistent Issues (Possible Defective Unit)

nisxarg, 2025-07-14:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

Installed via: sudo apt install ros-jazzy-turtlebot4-*  
Versions installed were the latest available as of July 2025.  

### Type of issue

Other

### Expected behaviour

Issues : System hardware/software instability – mapping, networking, and startup issues.

The robot should boot up quickly, connect reliably to Wi-Fi, pair with the Bluetooth controller, display correct battery status, and support stable mapping and navigation using official launch files.

### Actual behaviour

- Robot takes a long time to boot.
- Battery status shows 0%.
- Bluetooth controller does not connect.
- Communication light goes off randomly.
- Wi-Fi IP frequently shows "unknown" or disconnects.
- Repeated reboots are often needed to get partial functionality.
- Mapping launch reports many errors and instability.
- Navigation does not start properly after mapping.
- Connecting external display to Pi does not help — still unstable.
- Reflashing firmware did not resolve any of the above.

### Error messages

```bash

```

### To Reproduce

1. Power on the TurtleBot 4 (Standard).
2. Attempt to connect to a 2.4GHz or 5GHz Wi-Fi network.
3. Wait for IP to appear — sometimes shows "unknown."
4. Try pairing Bluetooth controller — fails.
5. Launch SLAM with: `ros2 launch turtlebot4_navigation slam.launch.py`
6. Mapping fails intermittently, reports errors.
7. Launch navigation: `ros2 launch turtlebot4_navigation navigation.launch.py`
8. Navigation fails to execute correctly.
9. Reboot and try again — same results.
10. Attempt firmware reflash — no change.

### Other notes

I've spent many hours troubleshooting and following all setup instructions closely. The issues are persistent and affect all core functions. I suspect the unit may be defective.

This GitHub issue is being submitted at the request of RobotShop to initiate the RMA confirmation process.

RB-ENantel, 2025-07-17:
Hi @RustyCPR 

This is Eric Nantel from RobotShop.
Can you have a look as this customer is requesting a return.

Regards,

smatarCPR, 2025-07-17:
Hello @nisxarg ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that your Turtlebot is not meeting expectations. With regarding to the issues you are facing:

- Robot takes a long time to boot. - Do you have the Turtlebot4 Diagnostics package enabled? If so, would it be possible to disable this package? 
- Battery status shows 0%. - What is the status of the COMM light on the top of the Turtlebot? If it is off, there is a break in connection between the Create3 and Raspberry Pi. If this is indeed the case, the battery display showing 0 is to be expected as the Raspberry Pi is not receiving data from the Create3, including battery status.
- Bluetooth controller does not connect. - For this would it be possible to use the p4pair script attached. to use the script run the commands:

 sudo apt install expect
chmod 755 p4pair
./p4pair <MAC of P4 Wireless Controller>

- Communication light goes off randomly/Wi-Fi IP frequently shows "unknown" or disconnects. - Are there additional network securities on the network you are working with? Are there multiple turtlebots on the same network? If there are multiple, you will need to use a discovery server configuration - https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#example-configuration

For the issues related to mapping and navigation, let's address the initial issues first to ensure that the Turtlebot is in a good state.

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

[p4pair (2).zip](https://github.com/user-attachments/files/21301338/p4pair.2.zip)