turtlebot/turtlebot4 #625: Map generation does not start

kadinche-sugizaki, 2025-08-12:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

Package: ros-jazzy-turtlebot4-navigation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 2472
Maintainer: Chris Iverach-Brereton <civerachb@clearpathrobotics.com>
Architecture: amd64
Version: 2.1.0-1noble.20250624.021150
Depends: ros-jazzy-nav2-bringup, ros-jazzy-nav2-simple-commander, ros-jazzy-slam-toolbox, ros-jazzy-ros-workspace
Description: Turtlebot4 Navigation

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

When RViz2 is launched, laser scan data should appear, SLAM should begin, and a map should be generated. Once a goal is set, the robot should begin navigating.

### Actual behaviour

Laser scan data is not displayed in RViz2, and map generation does not start. Even when a goal is set, the robot does not move. RViz2 status remains in “Planning” or “Idle.”

### Error messages

```bash
- ros2 launch turtlebot4_navigation slam.launch.py
[INFO] [launch]: All log files can be found below /home/geroika/.ros/log/2025-08-12-12-32-26-716291-Surface-5729
[INFO] [launch]: Default logging verbosity is set to INFO
[DEBUG] mapper_params_online_sync.yaml
[WARN] [1754969546.794343705] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[WARN] [1754969546.794392837] [rcl]: 'localhost_only' is disabled, 'automatic_discovery_range' and 'static_peers' will be used.
[INFO] [sync_slam_toolbox_node-1]: process started with pid [5742]
[sync_slam_toolbox_node-1] [WARN] [1754969547.144187343] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[sync_slam_toolbox_node-1] [WARN] [1754969547.144238906] [rcl]: 'localhost_only' is disabled, 'automatic_discovery_range' and 'static_peers' will be used.
[sync_slam_toolbox_node-1] [INFO] [1754969547.233029587] [slam_toolbox]: Node using stack size 40000000

- ros2 launch turtlebot4_viz view_navibation.launch.py
[INFO] [launch]: All log files can be found below /home/geroika/.ros/log/2025-08-12-12-34-22-335591-Surface-5803
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [5806]
[rviz2-1] [WARN] [1754969662.480374814] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
[rviz2-1] [WARN] [1754969662.480417324] [rcl]: 'localhost_only' is disabled, 'automatic_discovery_range' and 'static_peers' will be used.
[rviz2-1] [INFO] [1754969662.886160802] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1754969662.886400014] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-1] [INFO] [1754969662.947336448] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1754969663.046532067] [rviz2]: Subscribing to: /mobile_base/sensors/bumper_pointcloud
[rviz2-1]
[rviz2-1] [INFO] [1754969663.048572798] [rviz2]: Subscribing to: /mobile_base/sensors/bumper_pointcloud
[rviz2-1]
[rviz2-1] [INFO] [1754969663.073113011] [rviz2]: Subscribing to: /global_costmap/voxel_marked_cloud
[rviz2-1]
[rviz2-1] [INFO] [1754969663.109928211] [rviz2]: Subscribing to: /local_costmap/voxel_marked_cloud
[rviz2-1]
[rviz2-1] [INFO] [1754969663.158780757] [rviz2]: Subscribing to: /mobile_base/sensors/bumper_pointcloud
[rviz2-1]
[rviz2-1] [INFO] [1754969663.164270926] [rviz2]: Subscribing to: /global_costmap/voxel_marked_cloud
[rviz2-1]
[rviz2-1] [INFO] [1754969663.184398746] [rviz2]: Subscribing to: /local_costmap/voxel_marked_cloud
[rviz2-1]
[rviz2-1] [WARN] [1754969663.258021774] [rcl.logging_rosout]: Publisher already registered for node name: 'rviz2'. If this is due to multiple nodes with the same name then all logs for the logger named 'rviz2' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1754969663.295730307] [rcl.logging_rosout]: Publisher already registered for node name: 'rviz2'. If this is due to multiple nodes with the same name then all logs for the logger named 'rviz2' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [INFO] [1754969664.328582569] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969675.019739542] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969680.798793918] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969686.808562746] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969693.022700606] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969698.022950391] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969703.026651650] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969709.025884121] [rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1754969715.026879907] [rviz2]: Waiting for the slam_toolbox node configuration.
```

### To Reproduce

1. Launch SLAM and navigation in Terminal 1:
ros2 launch turtlebot4_navigation slam.launch.py

2. Launch RViz2 in Terminal 2:
ros2 launch turtlebot4_viz view_navigation.launch.py

3. RViz2 starts, but no laser scan is displayed and map generation does not begin.

[image]

[frames_2025-07-29_15.06.38.pdf](https://github.com/user-attachments/files/21726264/frames_2025-07-29_15.06.38.pdf)

### Other notes

It was working fine until last month, so it may have been updated.

smatarCPR, 2025-08-13:
Hello @kadinche-sugizaki ,
Thank you for contacting Clearpath support. I am very sorry to hear that your Turtlebot is not meeting expectations. Based on the details shared I have the following observations/recommendations:

- I notice in the output of slam.launch.py there is mention of ROS_LOCAL_HOST_ONLY being set: `[sync_slam_toolbox_node-1] [WARN] [1754969547.144187343] [rcl]: ROS_LOCALHOST_ONLY is deprecated but still honored if it is enabled. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.`
- Is this being set on your remote computer? This would be in either the ~/.bashrc file or a setup.bash file. This feature is no longer used in ROS 2 Jazzy. Would it be possible to unset this variable and see if the issue is cleared?
- Is ROS super client set on the terminal launching the navigation stack?

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

kadinche-sugizaki, 2025-08-14:
Thanks for your guidance. @smatarCPR 
I found that `ROS_AUTOMATIC_DISCOVERY_RANGE` was previously set to `LOCALHOST` on my remote PC. After updating it to `SUBNET`, RViz began to behave differently.

Specifically:
- `GlobalStatus` and `RobotModel` now show `Error` instead of `Warn`, which suggests that RViz is receiving data, though something may still be misconfigured.
- In RViz's GlobalOptions, the `Fixed Frame` dropdown no longer includes `map` as an option.

This makes me suspect that the `map` frame isn't being published or discovered properly. I’ll investigate the TF tree and relevant nodes next, but if you have any suggestions on what might be missing or misconfigured, I’d appreciate your input.

[image]

smatarCPR, 2025-09-18:
Hello @kadinche-sugizaki 
Apologies for the delayed response on this. Thank you for providing the requested details. Based on the details shared I have the following questions/requests:

- With the new setup would it be possible to provide the updated terminal outputs? Some of the details may have changed highlighting a new/different issues that may be causing the map generation not to start.
- In addition to the new logs, for each terminal would it be possible to provide the output of the following command - `printenv | grep ROS`

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif