turtlebot/turtlebot4 #626: Turtlebot4 is not undocking

braib, 2025-08-13:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I believe it is latest

### Type of issue

Battery

### Expected behaviour

I want the robot to undock when i press undock button in  my robot.

### Actual behaviour

The robot is not undocking

### Error messages

```bash
no error message
```

### To Reproduce

1. press undock button i robot
2. the robot is not undocking

### Other notes

tb4 is not undocking, also when i see the battery percentage is showing 0 in robot , but in website it is 99. also i dont have any topic or service related to undock or dock. 
_Note: I have added namespace_

smatarCPR, 2025-08-13:
Hello @braib ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that your Turtlebot is not meeting expectations. Based on the details shared I have the following questions/recommendations:

- Is the Create3 connected to a wi-fi network? If so, was it assigned an IP address and are you able to view the IP address on the webserver? https://iroboteducation.github.io/create3_docs/webserver/connect/
- What is the status of the top lights on the Turtlebot 4?
- If you remove the namespace from the ROS setup details in the Turtlebot 4 setup utility is the Turtlebot 4 able to dock and undock? 
- If you switch to a discovery server setup, does the behaviour of the Turtlebot change?
- Are you able to echo the Create3 specific topics such as /odom when SSH'ed into the Raspberry Pi?

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2025-10-01:
Hello @braib 
Apologies for the delayed response on this matter. I have the following recommendations/observations based on the details shared:

- Typically when the battery display is set to 0, this is an indication that the Raspberry Pi and Create 3 units are unable to communicate with each other. This will be accompanied with the BATT and COMM lights being off.
- As you are using a simple discovery setup I have the following questions regarding the network you are using: Is the Create3 connected to a wi-fi network? If so, was it assigned an IP address and are you able to view the IP address on the webserver? https://iroboteducation.github.io/create3_docs/webserver/connect/
- What is the bandwidth of the network you are using is it a 5.0 GHz network or a 2.4 GHz network?
- Is there a great deal of network traffic? (i.e. other robots on the same network?)
- Are you able to switch to a discovery server setup?

Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2025-11-06 by smatarCPR