turtlebot/turtlebot4 #627: TurtleBot4 (RB-Cle-03) Power Issue – Raspberry Pi & Sensors Not Receiving Sustained Charge

kushtoby, 2025-08-15:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

I do not know

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

ii  ros-jazzy-turtlebot4-description                 2.1.0-1noble.20250624.063239             amd64        Turtlebot4 Description package
ii  ros-jazzy-turtlebot4-desktop                     2.0.1-1noble.20250624.064100             amd64        Turtlebot4 Desktop Metapackage
ii  ros-jazzy-turtlebot4-viz                         2.0.1-1noble.20250624.063625             amd64        Visualization launchers and helpers for Turtlebot4

### Type of issue

Battery

### Expected behaviour

When powering on the TurtleBot4, the Raspberry Pi and sensors should receive a stable charge and operate continuously.

### Actual behaviour

While the TurtleBot4 powers on, the Raspberry Pi and sensors do not receive sustained power, causing shutdowns or intermittent operation. Also, the turtlebot turns off immediately it is undocked.

### Error messages

```bash
Screen doesn't power up at all either.
```

### To Reproduce

Power on TurtleBot4 (RB-Cle-03)
Observe Raspberry Pi and sensors
Devices fail to maintain operation or shut down intermittently

### Other notes

The TurtleBot4 has seen very limited use since acquisition in May 2023.
Unable to locate a compatible replacement battery through the Clearpath website or other vendors.
Seeking guidance for troubleshooting power supply issues and obtaining an official replacement battery if needed.

kushtoby, 2025-08-19:
Hi, thanks. I tried both solutions and neither worked. Also, the battery is currently unavailable on the website. 

This also happens just before the robot powers on when docked. And then, it goes back to white on its own.
https://github.com/user-attachments/assets/920fc121-a74c-48a1-af95-cb97ba628368

RustyCPR, 2025-08-19:
Hi there @kushtoby 
Thanks for the video - it seems like the battery is very low. 
See this iRobot Create3 page for light ring color definitions: https://iroboteducation.github.io/create3_docs/hw/face/

I searched Amazon.com for "ABL-D1 Battery for iRobot" and it came back with many options. 

Best regards,
Rusty