turtlebot/turtlebot4 #628: Not Working New out of Box Turtlebot4

tsatharasi, 2025-08-25:
### Robot Model

Turtlebot4 Lite

### ROS distro

Galactic

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Installed with the robot. It is new just opened from box.

### Type of issue

Networking

### Expected behaviour

The Bot should be Moving, which it shd not. The test cases shd be passes, none pass. Also observed that the light tests dont pass. The robot is new out of box.

### Actual behaviour

The Bot neither does move, also the test cases fail in turtlebot4-setup

### Error messages

```bash
No Error Messages. There is no netwrok or component detected at USB from the raspberry pi, which suggests the microros controller is not connecting to the raspi board.
```

### To Reproduce

1. use joystick or teleop-twist-keyboard to move the robot, it doesn't move.

RustyCPR, 2025-08-27:
Hi there @tsatharasi 
Sorry to hear your experiencing concerns with your TB4 Lite. 
Could you please update the OS to a newer supported version - we have a YouTube step by step tutorial video here to update to ROS2 Jazzy: https://youtu.be/VsmAVYNyYQs?si=i7MtDZ_ciiiLDrww
Thanks,
Rusty

tsatharasi, 2025-08-27:
Hey @RustyCPR,

The Bot was shipped in and runs ROS2 Jazzy. Based on my troubleshooting I think the USB data connection is having issues and hence the bot doesn’t communicate with the create3 controller

TS

[External Email]

RustyCPR, 2025-08-27:
Hi @tsatharasi glad you were able to find and resolve your issue. 
I will go ahead and close this ticket. Should you need further assistance, please open a new ticket. 
Kind regards,
Rusty