turtlebot/turtlebot4 #630: TB4 Standard can't move forward, only spin on itself

ValentinLeglise, 2025-09-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Select One

### Built from source or installed?

Installed

### Package version

vleglise@pc-front:~$ dpkg -s ros-humble-turtlebot4-desktop
Package: ros-humble-turtlebot4-desktop
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 1.0.0-1jammy.20250728.103520
Depends: ros-humble-turtlebot4-viz, ros-humble-ros-workspace
Description: Turtlebot4 Desktop Metapackage

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

When teleoperating the TB4 forward/backward, with teleop_twist_ keyboard or with ros2 run pub /cmd_vel, the robot move forward/backward.

### Actual behaviour

My customer bought 14 TB4 for his students. He has set them all up by following [basic setup guide](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html), but only one has this problem. After basic set up, my customer can see all ros2 topics through WiFi and try to teleoperate the robot with node teleop_twist_keyboard or with pub /cmd_vel.

In both case, when he tries to make the robot move forward (using "i" with teleop_twist for example), the robot only turn around itself randomly.
I have added a video showing the behavior of the robot, when my customer only press the key "i" during teleoparation.

So I received his robot to make tests, I have reflashed the RPi, connect it to my network and the problem still persist.

I think it is a hardware problem because all the 13 others robots are working well, and they all have the same configuration as the defective one.

Can you help ?

https://drive.google.com/file/d/1ZR4BBbpHITywc-JRuAtRsppxyRlxhryu/view

### Error messages

```bash

```

### To Reproduce

1. Follow basic setup with simple discovery network configuration
2. Teleoperate the robot

ValentinLeglise, 2025-09-08:
Hello ClearPath.

Sorry to bother you. Is it possible to have an answer for this ?
It is quite urgent for my customer to get back his robot.

Best regards, Valentin.

smatarCPR, 2025-09-08:
Hello @ValentinLeglise 
Thank you for reaching out on the Turtlebot 4 github page and apologies for the delayed response. I am very sorry to hear that this Turtlebot is not meeting expectations. I have the following requests regarding the current state of the Turtlebot:

- If possible, are you able to attempt a factory reset of the Turtlebot, upgrading to ROS 2 Jazzy. This will also require flashing the firmware on the Create3 to a newer version.
- Should the issue persist following the upgrade, would it be possible to provide the Create3 log files, and the Turtlebot using `journalctl -b > turtlebot4_log.txt`
- Would it also be possible to provide a ROS bag recording of the Turtlebot being commanded to drive straight using a ros2 cmd_vel topic. Please capture all topics in the rosbag using the -a flag.

Looking forward to hearing from you soon.
Best Regards,
Saif M.

ValentinLeglise, 2025-09-10:
Hello,

Thank you for your response.

You can see below the logs of the create3 while I was operating the robot in SSH with the command ros2 topic pub /cmd_vel.
When I tried to teleoperate the robot with teleop_twist_keyboard remotely from my pc, it didn't work. Only command in SSH seems to work, with a pretty long delay.

[messages.txt](https://github.com/user-attachments/files/22256452/messages.txt)