turtlebot/turtlebot4 #633: Return Requested: TurtleBot4

t-satharasi, 2025-09-15:
### Robot Model

Turtlebot4 Lite

### ROS distro

Jazzy

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

The Turtlebot4 fails to move with joy or keyboard, also from within the Raspberry Pi. Tried updating the Create3 firmware. Followed the Troubleshoot Guide from a previous ticket. Still fails to move. After the initial Setup, the test functions from turtlebot4-setup do not work, neither the light tests nor any other test. After further investigating, it was found that the bot is not connected to the controller and hence does not move. This might be due to damaged hardware; we are insisting on a refund for the Turtlebot 4 robot. Please let us know the steps required.

Please let us know.

We have tried both the Simple and Discovery Server as part of our Laboratory, through which other Robots Work great at the University of Florida, FL 

The Robot was purchased from Robot Shop and was shipped to the University with ROS2 Jazzy, Raspberry Pi 4 Compute Board, and Ubuntu Server 22.04. 

### Type of issue

Other

### Expected behaviour

The Tests are working, the robot is moving, etc

### Actual behaviour

The tests didn't work, nor is the robot moving, docking, undocking, etc.

### Error messages

```bash
Does not apply.
```

### To Reproduce

1. Order a Turtlebot4 Lite from RobotShop USA
2. Switch it on after opening the box
3. Try the initial setup as indicated on the website
4. Go to Tests and Light Tests or Joystick

### Other notes

Tried both Discovery and Simple Server, but neither works.
The USB shows no device connected, even with the Create 3 USB connected and powering the Compute Board.

t-satharasi, 2025-10-01:
1. What is the Create3 base power button light color and pattern? 
white, solid pattern, does not change unless 

2. Are you keeping the namespace as _do_not_use? 
Yes.

3. During the setup are you attempting to setup a simple discovery setup or a discovery server setup?
Tried both didn't work!.

4. Are there other robots on the network?
In the Network Yes:
There are a couple of bebop parrots and husarion rosbots, which work fine.

5. Which firmware did you load on the Create3?
I.264

If setting up a simple discovery:

6. Is the network you are working with a dual-band network (i.e., both 2.4 GHz and 5.0 GHz are available)? If it is a single band, which frequency is being used? The create3 only has the capability of connecting to a 2.4 GHz network. If the network separates the 5 GHz and 2.4 GHz networks onto separate SSIDs, are they bridged in the background?
They bridge in the background. They have 5G and 2.4, but the robot and Create3 connect to 2.4 Hz Wifi.
 
9. After setting up the Create3 to connect to wi-fi network, are you able to connect to the webserver, or does it only work when in Access Point mode?
It does connect to the Web server every time.

10. Are you able to factory reset the Create3? - https://iroboteducation.github.io/create3_docs/webserver/about/#factory-resetf setting up discovery server:9. Is the create3 attempting to connect to a network? During a discovery server setup it should not be connecting to a network.

No, I was not able to reset.

Can you please share logs:

I shall share them shortly.

RustyCPR, 2025-10-30:
Hi there @t-satharasi 
Your ROS version is detailed as ROS2 Jazzy
OS is listed as Ubuntu 22.04 which would is compatible with ROS2 Humble not Jazzy
Your Create3 firmware is Iron version I.264 - which is not listed on iRobots Create3 firmware site (https://iroboteducation.github.io/create3_docs/releases/overview/)

Seems like you have mismatched OS / ROS versions and Create3 firmware. 
Please perform a factory reset and set up one Ubuntu/ ROS2 / Firmware set up. 

Kind regards,
Rusty

t-satharasi, 2025-11-05:
Please send a link with the detailed instructions.
These were not installed in any such manner. They came in shipped into them.

Thanks and Regards
TS

RustyCPR, 2025-12-18:
Hi there @t-satharasi I replied to your email.
Will support you via email.

Thanks,
Rusty