turtlebot/turtlebot4 #637: Poor Battery Performance

callum-sh, 2025-09-17:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

all

### Type of issue

Battery

### Expected behaviour

Operation time 	2h 30m - 4h depending on load

https://turtlebot.github.io/turtlebot4-user-manual/overview/features.html

### Actual behaviour

Getting < 1hr of operating time between charges making any and all work quite challenging

### Error messages

```bash

```

### To Reproduce

1. run robot for a while and query nodes through scripts 

### Other notes

[journal.log](https://github.com/user-attachments/files/22390024/journal.log)
[create3.log](https://github.com/user-attachments/files/22390025/create3.log)

RustyCPR, 2025-09-18:
Hi @callum-sh 
Thanks for reaching out regarding your TurtleBot 4’s battery life.
How old is your TurtleBot 4 is? That will help us better understand whether battery age may be a factor.
In the meantime, our team suggested a few steps that may help extend runtime:

- If you’re running SLAM but not using the OAK-D camera, try disconnecting it to reduce power draw.
- Offloading heavy processes from the Raspberry Pi to your remote PC can help reduce onboard power consumption.
- When mapping large facilities, optimizing your SLAM parameters can improve efficiency.
- As mentioned in [ticket 518](https://github.com/turtlebot/turtlebot4/issues/581#issuecomment-3300300855) , you may also consider using a 12V external USB hub and battery pack to help reduce consumption. This would connect to the RPi and supply power to the sensors. 
- That said, depending on the scale of your project, you may find that a UGV platform with greater battery capacity and onboard processing power would be a better long-term fit. Example: https://clearpathrobotics.com/dingo-indoor-mobile-robot/

Best regards,
Rusty

RustyCPR, 2025-09-22:
Hi @callum-sh 
How old is your TurtleBot 4 is? That will help us better understand whether battery age may be a factor.
> If over a year old - you can try sourcing another battery - Amazon.com search: _ABL-D1 Battery for iRobot_

**do you have recommended SLAM settings or any documentation to read more?**
> See: https://github.com/SteveMacenski/slam_toolbox?tab=readme-ov-file#slam-toolbox
>Pay attention to minimum_travel_distance 
>https://docs.nav2.org/development_guides/involvement_docs/index.html#id1 

Best regards,
Rusty