turtlebot/turtlebot4 #639: Reliability Issues

callum-sh, 2025-09-24:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

all

### Type of issue

Other

### Expected behaviour

Hoping that once the robot is running, and Nav2 is working, that nodes stay consistently publishing and will continue taking commands from devices on the network. 

### Actual behaviour

After running for a prolonged period of time, the robot stops publishing nodes (sometimes ip nodes, sometimes navigation nodes, etc.) and to fix this, we need to restart the robot.

Let me know what information is needed to help make the system more reliable 

### Error messages

```bash

```

### To Reproduce

1. start robot as per documentation
2. run robot for 1+ hours

RustyCPR, 2025-10-23:
Hi there @callum-sh 
Thanks for writing in and sorry your having issues. 

First, confirm what exactly stops publishing? 
1. Does `rqt_graph` or `ros2 node list` still show active nodes, or is everything frozen?
2. Can you capture the below details before / after the occurrence? 

```
- ros2 topic list
- ros2 node list
- ros2 topic hz /odom
- ros2 topic hz /scan
```

3. Can you share if the RPi4 is overloaded or throttled after occurrence? `htop`
4. Send output of `dmesg` after occurrence. 
5. Logs:

- Create3 Logs: `https://iroboteducation.github.io/create3_docs/webserver/logs/`
- `sudo journalctl -u turtlebot4 -r`

Thanks,
Rusty

RustyCPR, 2025-12-04:
Hi @callum-sh I'll close this ticket due to in-activity. 
If you require further support please open a new ticket. 

Thank you,
Rusty